FreddieHong19/Open5x #5: Consider porting to BTT GTR

pyr0ball, 2022-03-24:
It seems TeamGloomy have gotten a decent port of RRF to work on these cheaper GTR boards. Would you be willing to consider building a version specific to that setup, or assist me in porting it?

I plan on re-engineering this project for an old FlashForge Creator Pro, and happen to have one of these GTR's on hand to work with

https://teamgloomy.github.io/gtr_pins.html

Keybored02, 2022-03-30:
I second this. The BTT GTR, SKR (with the 3axis expansion module), and Octopus are extremely widespread among makers. The Duet boards, given the cost, are more for prosumers. A Marlin port would also be a great improvement, given how 2.0.9 has simplified the addition of more (linear) axis.

FreddieHong19, 2022-03-31:
Hi, 
Thank you for sharing this information about your GTR board suggestion! I will have a read through their documentation and see if it could be implemented. Personally for now, I am very satisfied with Duet board especially regarding their hardware, expandability and their web controller. I

pyr0ball, 2022-03-31:
If I had to choose between the two and price was no object, (or if I had one on hand myself) I'd probably feel the same!

Do you think if I simply used the configs from this project on the existing RRF fork for the GTR, it might just work? I'd be happy to give it a shot if so. If there's any obvious pitfalls with that method, and you could point them out ahead of time, I'd be happy to do the legwork on investigation and testing

closed 2022-10-18 by FreddieHong19