FreddieHong19/Open5x #7: Question about model slicing

Shuwrood-SSW, 2022-06-01:
Thanks for sharing your great works, but I still have no idea much about the slicing software. 
For examole, when I use the model of Supportless_sample_2 (small fan), the otary bearing can't be fixed and positioned. I would appreciate a detail video tutorial  about slicing if you could.  Moreover,  can you upload the model of the spiral staircase used in your video for better. Thank you very much.

FreddieHong19, 2022-06-05:
Hi Shuwrood,

From your screenshot it appears that the model which you are importing is not at 0,0,0 coordinates. If you position the bottom face of the model, the slicing preview may appear at the right location.
I am going through iterations of development to create a stable version of the slicer +I am planning more detailed video tutorials. I am hoping to share them in a nearer time, but it is taking longer than I have anticipated! 
Best,

Shuwrood-SSW, 2022-06-06:
That's really great. Looking forward to the video tutorial! Besides, could you upload the following model to github.

[image]

syalways, 2022-07-14:
sorry,I have a question that this model can not  print in_fill lite,is itright?