FreddieHong19/Open5x #24: [FEATURE REQUEST] Add release

darigovresearch, 2022-11-09:
Would be nice to have a V1.0.0 release in GitHub to make it clearer to see when there are major & minor releases.

Instructions on how to do this can be seen [here](https://docs.github.com/en/repositories/releasing-projects-on-github/managing-releases-in-a-repository)

Happy to do this if you wish just need to be given admin or maintainer rights if you're happy to do so

FreddieHong19, 2022-11-09:
Hi Darigovresearch,

Thank you for the suggestion! 
There are some small updates remaining in this version which should be uploaded soon. I will release V1.0.0 after.

darigovresearch, 2022-11-09:
Hi Freddie,

You're very welcome & thanks for the update.
Don't hesitate to release a V0.1.0 already and you can add some small updates as v0.2.0 etc as the semantic versioning describes in the repository