FreddieHong19/Open5x #33: Make available for Mac Rhino Users

mikedattolo, 2022-12-12:
I use a mac and can not open the grasshopper file on Rhino because the plugin used to make the file is only useable on Windows machines. Is it possible to make this available in a way mac users can see and use it as well...

BigKoala33, 2022-12-14:
Hi @mikedattolo what is the name of that plugin?

BigKoala33, 2022-12-14:
Hi @mikedattolo what is the name of the plugin?

mikedattolo, 2022-12-15:
That's the problem. I'm not sure. I believe they are both Pufferfish and Heteroptera; however, I don't think it is a direct issue having to do with them. It has to do with something referenced in the grasshopper file here 
[image]

mikedattolo, 2022-12-15:
And sorry again, I forgot that I am trying to use or make my own simulation through grasshopper to slice G-code for a 4-5 axis printer prototype I am working on for a class in college. It doesn't matter if I can't do it, but I would like to because I think this as a research subject/innovation for 3d printing would be amazing.

mikedattolo, 2022-12-16:
@BigKoala33