FreddieHong19/Open5x #36: Toolpath generation

walterbux, 2022-12-16:
Hello Freddie,
thank you for your wonderful work.
I cannot find how to run this part of the conformal slicer (see attached picture).
Can you help me please ?

Walter 
[image]

xuanxuanbingo, 2023-03-21:
Hello, I want to ask have you successfully found this part of the conformal slicer ? I also need help

walterbux, 2023-03-21:
Unfortunately no, I have not found this part. I have tried to use the file 2022_03_22_open5x_supportless_surface_Lite.gh but it always gives me errors or warnings.

FreddieHong19, 2023-03-21:
Hi Walterbux,

sorry for my late response, the above example is previous version of the definition which uses (Human UI) library. The new iteration shared on GitHub is no longer using the Human UI library.
I have been working on the grasshopper definition and I am planning to share the new version sometime in April, and this time with instruction video too.

walterbux, 2023-03-21:
OK, thank you i am waiting impatiently...

FreddieHong19, 2023-03-22:
Hi,
I will post a very clear step-by-step video in April. I am in my last few weeks of PhD, handling various things at once.
Thank you for the patience!

xuanxuanbingo, 2023-04-08:
Hi,
It's the step-by-step video ready? Now our project is in urgent need of your help with this video. Really thank you for your early reply.