FreddieHong19/Open5x #42: PufferFish plugin not able to download

mikedattolo, 2023-03-15:
I try and open up the grasshopper file and I am able to download the first plugin no problem, However when it asks me to install pufferfish it fails, I have tried going to food4Rhino and manually downloading it but it doesn't acknowledge it and still asks me to download it and when I close out of the prompt I get a couple errors. At the same time I was wondering if this can be repurposed for other types of 5D printers ones that specifically uses rotational axis on the x-axis rather than the bed of the printer.

FreddieHong19, 2023-03-21:
Hi,
the grasshopper definition should still be functional without the pufferfish library. (Could you post a screenshot of an error)
The new version of the grasshoper definition will be shared soon during april!

mikedattolo, 2023-03-21:
> Hi, the grasshopper definition should still be functional without the pufferfish library. (Could you post a screenshot of an error) The new version of the grasshoper definition will be shared soon during april!

hey, thank you for responding. I figured it out. It was just because I had the pufferfish plugin block, and now it works, however I still am having issues with actually trying out the slicer, when I add in the brep and connect it to the supportless brep and substrate the slicer cant find the path curve that is connected to the conformal pathing cluster.

closed 2023-03-24 by mikedattolo