FreddieHong19/Open5x #53: using Russian 5D printer as a prototype

Divertisment, 2023-06-17:
I propose to use in development and steal and decompile what is not publicly available from this Russian 5d printer. 
They have a successfully working and commercially available printer in two variants. 
https://5dtech.pro/5d-530-fiber/

Here an open source slicer, it seems to me
https://github.com/stereotech/STE-Slicer

the control software (STE App) is a reworked fork the Cure to which they added a 5d processing module and made it proprietary (in violation of the license obviously)

sotnikovmisa, 2023-10-08:
how are the results?

sotnikovmisa, 2023-10-08:
It’s not easier to use a cracked Rhinoceros

vabka, 2023-11-10:
Let's violate license agreement of commercial software and use it's code in open source. What can go wrong? Lol

dimensionalcreations3d, 2023-11-14:
They want 18,500 USD for their fiber printer.

Seems like they're over estimating the value of what they built. 

It's a cool machine, but there's enough open source gear out there to build one for far less.

Slicing is gonna be the main issue until it's "perfected".

That being said, I don't think stealing someone else's work is ever a good idea. It doesn't seem anything they have is proprietary (legally speaking) beyond their assertion that it is.